hamishcoleman/thinkpad-ec #223: make patched.t530.img not working

Schubsi, 2022-08-25:
When I try to create the patched image i always get the following:

[image]

It might have something to do with the dependencies, but im not sure.
When i type in "sudo make build-deps", I get back "make: *** No rule to make target 'build-deps'.  Stop."

I dont know where to go from here, help is greatly appreciated.

hamishcoleman, 2022-08-25:
The error `No rule to make target build-deps`, just highlights that I made a trivial mistake in the instructions.

If you run things in a slightly different order - after `cd ~/thinkpad-ec` then run the `make build-deps`

I'll just go and fix the instructions - sorry.

Schubsi, 2022-08-25:
Thanks for the quick answer!
I think I know now what the problem was.
When installing the prerequisite packages some packages, namely "xorriso" , "innoextract" and "mtools" were missing.
I fixed this by going to "software & updates" and enabling the first 4 options and then running make build-deps again.
It seems like the Image was created successfully now, ill try flashing it and will report back later.

Schubsi, 2022-08-25:
Bios installed without a problem and patches work, thank you very much! 😄

closed 2022-08-25 by Schubsi

hamishcoleman, 2022-08-25:
However, I have no real idea what "the first 4 options" in this software and updates tool might be - I dont use that - can you give me some more details?  Are they different repository subsections?

Schubsi, 2022-08-25:
I used [this](https://askubuntu.com/questions/378558/unable-to-locate-package-while-trying-to-install-packages-with-apt) guide.
Im on Ubuntu 22.04. When you click the bottom right corner a screen opens up where you can search for all installed applications. I opened "software & updates" and made sure to enable the first 4 options under the "Ubuntu Software" Tab.
[image]

This probably doesn't sound very technical, but im very new to ubuntu/linux.

hamishcoleman, 2022-08-25:
Thanks! It might not be very technical, but you were very clear.  It's enough detail that I can work out what packages were where.