hamishcoleman/thinkpad-ec #224: Support for ThinkPad Edge E530

Gamemaster2930, 2022-08-30:
Hello, is it possible to add support for ThinkPad Edge E530? I would like to patch my ThinkPad because my replace battery is not working (whitelist message on boot => not charging) but it is not on the list... Thank you

Gamemaster2930, 2022-09-10:
Can someone possibly answer this question? I need it quite fast....

sina-devel, 2022-10-18:
I have the same request

maexl0105, 2022-11-23:
Same Request

KudryavtsevAlex, 2023-01-19:
Me too, have the same request.
What can we do to create, to make the battery patch?

sedative82, 2024-05-11:
I have tha same problem too.

porteusconf, 2026-02-15:
I think all we want is battery whitelist removal. Can anyone outline the steps to create a modded bios that just removes the battery check? For example, what tools do I install on debian (`apt install`) to dump my existing bios and mod it?  Sometimes I can get my E530 to boot without complaining about the battery by swapping batteries, holding down power button for 60 sec with no battery/no power, etc.   But the darn issue keeps coming back: "battery not suported... may not charge". 

Note: I was using tpm to set charge limits (aka battery conservation mode). I've tried `tpm fullcharge` and `sudo systemctl disable tlp.service` and sometime battery will charge after reboot. I disabled tlp since I have enabled in gnome: settings: power: Preserve Battery Health.
BTW, all the batteries I am trying are oem Lenovo batteries that originally came with these old laptops, but laptop were only used a few months of the year their entire lifetime.  

```
Info below on bios of E530 is from lshw-gtk in debian trixie...
$ uname -a
Linux EdgeE530 6.12.69+deb13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.12.69-1 (2026-02-08) x86_64 GNU/Linux

version: ThinkPad Edge E530
product: 3259CTO (SKU)  

        width: 64 bits capabilities: SMBIOS version 2.7, DMI 
	version 2.7,LENOVO version: H0ET32WW (1.14 ) size: 128KiB 
	capacity: 4MiB capabilities: PCI bus, Plug-and-Play, BIOS EEPROM 
	can be upgraded, BIOS shadowing, Booting from CD-ROM/DVD, 
	Selectable boot path, Enhanced Disk Drive extensions, 3.5" 720KB 
	floppy, Print Screen key, i8042 keyboard controller, INT14 
	serial line control, INT17 printer control, INT10 CGA/Mono 
	video, ACPI, USB legacy emulation, BIOS boot specification, UEFI 
	specification is supported Symmetric Multi-Processing, 32-bit 
	processes configuration: 
        administrator_password: disabled chassis: 
        notebook family: ThinkPad Edge E530
	power-on_password: disabled

BIOS 
/0/d    vendor: LENOVO version: H0ET32WW (1.14 ) size: 128KiB capacity: 4MiB 
        capabilities:
	PCI bus, Plug-and-Play, BIOS EEPROM can be upgraded, BIOS 
	shadowing, Booting from CD-ROM/DVD, Selectable boot path, 
	Enhanced Disk Drive extensions, 3.5" 720KB floppy, Print Screen 
	key, i8042 keyboard controller, INT14 serial line control, INT17 
	printer control, INT10 CGA/Mono video, ACPI, USB legacy 
	emulation, BIOS boot specification, UEFI specification is 
	supported
```