hamishcoleman/thinkpad-ec #227: b590 ERROR 081 - FAILED TO LOAD BIOS IMAGE FILE! sTATUS = 101

deeptenk2, 2022-11-29:
Hello. b590 ERROR 081 - FAILED TO LOAD BIOS IMAGE FILE! STATUS = 101
? 
https://ibb.co/7zG51ns

leecher1337, 2022-11-29:
Hm, hard to tell, I don't have a B590 here for testing. The error 101 might indicate that the file to be patched may not even exist on the source medium. Have you tried to check with `dir` command if it's there? I didn't see the commandline for the flasher in the photo, but that's the first thing I would check: Does the BIOS update file exist in the image?

As EC hotpatching ISO works, the code change itself must be correct. 

As you asked me via e-mail why my ISO doesn't allow permanent flashing: The reason is already explained in the project's readme:  https://github.com/leecher1337/thinkpad-Lx30-ec#patch-and-reflash-firmware
So the flash chip cannot be unlocked, except with some hackery that 1vyrain could do, but that would be more overhead than just flashing with original flash procedure what thinkpad-ec is supposed to do.

If you patched the image manually, did you ensure to also update the checksum of the embedded controller ( https://github.com/leecher1337/thinkpad-Lx30-ec/blob/main/fwpat/util/npce885crc.c )? Otherwise your updated firmware won't load and you can only (possibly) recover using a flash programmer, so take care!

deeptenk2, 2022-11-30:
Thanks for the answer. Sorry, my English is bad. I did not calculate the checksum, you are right, but I do not quite understand how to do it with *.bin File created by the programmer (CH341a programmer). The file is present, but even if you replace it in the original flash driver and the flash driver under Windows by Lenovo do not see the file, I checked the path, the file is in place. Today I will check through the iso file and CD and report the result.

deeptenk2, 2022-11-30:
https://ibb.co/BqGKphW screen 
These are the images I got.
https://ufile.io/wisi2xa7

deeptenk2, 2022-11-30:
https://ibb.co/BqGKphW screen by ISO by DVD-RomThese are the images I got.https://ufile.io/wisi2xa7 -- С уважением,Святослав ***@***.***

deeptenk2, 2022-11-30:
If I send you the bin file read by the programmer https://ufile.io/xj8d2vnj, can you modify it correctly or tell you step by step how to do it?

leecher1337, 2022-12-01:
Hi,

The thinkpad-ec repository here is buggy. As mentioned in my original thread, I don't recommend using it and instead use my original repository, as the repository here doesn't seem to be tested very well.
So, please use https://github.com/leecher1337/thinkpad-ec/tree/Lx30 instead.

If you diff the output of the building process, you can see that this repository uses a wrong filename when writing to the disk image causing mentioned failure that file wasn't found:

< = this repostiroy = wrong
> = my repository = correct
```
< cp --reflink=auto b590.H9ET92WW.CAP.orig b590.H9ET92WW.CAP.tmp
< ./scripts/FL2_copyIMG to_fl2 b590.H9ET92WW.CAP.tmp b590.H9ET92WW.img
< IMG at offset 0x4001d0 size 0x20000 (EFI::Capsule::PFH_header b590.H9ET92WW.CAP.
                                 tmp)
< mv b590.H9ET92WW.CAP.tmp b590.H9ET92WW.CAP
---
> cp --reflink=auto H9ET92WW.CAP.orig H9ET92WW.CAP.tmp
> ./scripts/FL2_copyIMG to_fl2 H9ET92WW.CAP.tmp b590.H9ET92WW.img
> IMG at offset 0x4001d0 size 0x20000 (EFI::Capsule::PFH_header H9ET92WW.CAP.tmp)
> mv H9ET92WW.CAP.tmp H9ET92WW.CAP
```

Then you should receive a correct .iso file that you can boot and patch. If you are still having problems, I can also patch your BIOS dump, but I don't think it will be necessary.

deeptenk2, 2022-12-01:
Я попробовал, результат тот же. Как под Windows так и под dos. Размер файлов также отличается.
Я все же попрошу вас сделать мне рабочий *.img и *.bin для программатора. Заранее благодарю.
https://ibb.co/0mvz92T

`deep@deep-virtual-machine:~/thinkpad-ec-Lx30$ make patched.b590.img
Generated dependancies from descriptions
Downloading b590,b490 BIOS ?.? (H9ET92WW) EC 1.02 (H9EC09WW)
2022-12-01 09:14:54 URL:https://download.lenovo.com/pccbbs/mobiles/h9et92ww.exe [4525327/4525327] -> "h9et92ww.exe.orig" [1]
scripts/checksum --mv_on_fail h9et92ww.exe.orig h9et92ww.exe
mv `innoextract h9et92ww.exe.orig -I H9ET92WW.CAP | grep -i H9ET92WW.CAP | cut -d'"' -f2` H9ET92WW.CAP.orig
./scripts/FL2_copyIMG from_fl2 H9ET92WW.CAP.orig b590.H9ET92WW.img.orig
IMG at offset 0x4001d0 size 0x20000 (EFI::Capsule::PFH_header H9ET92WW.CAP.orig)
./scripts/hexpatch.pl --rm_on_fail --fail_on_missing --report b590.H9ET92WW.img.report b590.H9ET92WW.img b590.H9ET92WW.img.d/006_battery_validate.patch
Attempting to patch b590.H9ET92WW.img
Applying b590.H9ET92WW.img.d/006_battery_validate.patch -4178,14 +4178,14
mkdir nuvoton-tools
wget -O nuvoton-tools/npce885crc.c https://raw.githubusercontent.com/leecher1337/thinkpad-Lx30-ec/main/fwpat/util/npce885crc.c
--2022-12-01 09:14:55--  https://raw.githubusercontent.com/leecher1337/thinkpad-Lx30-ec/main/fwpat/util/npce885crc.c
Распознаётся raw.githubusercontent.com (raw.githubusercontent.com)… 185.199.109.133, 185.199.110.133, 185.199.111.133, ...
Подключение к raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... соединение установлено.
HTTP-запрос отправлен. Ожидание ответа… 200 OK
Длина: 3276 (3,2K) [text/plain]
Сохранение в: ‘nuvoton-tools/npce885crc.c’

nuvoton-tools/npce8 100%[===================>]   3,20K  --.-KB/s    за 0,1s    

2022-12-01 09:14:56 (27,6 KB/s) - ‘nuvoton-tools/npce885crc.c’ сохранён [3276/3276]

gcc -o nuvoton-tools/npce885crc nuvoton-tools/npce885crc.c
fatal: не найден git репозиторий (или один из родительских каталогов): .git
./nuvoton-tools/npce885crc -o 0x8000 -u b590.H9ET92WW.img
NUVOTON Insyde firmware CRC calculator v1.00
(c) leecher@dose.0wnz.at 02/2022

Searching at offset 8000
CRC Start address=0048800A
Assumed load address: 00488000
CRC indexing offset = A
Calculated CRC = A2, file CRC = 75 --> CRC differs!
Updating CRC...OK!
cp --reflink=auto H9ET92WW.CAP.orig H9ET92WW.CAP.tmp
./scripts/FL2_copyIMG to_fl2 H9ET92WW.CAP.tmp b590.H9ET92WW.img
IMG at offset 0x4001d0 size 0x20000 (EFI::Capsule::PFH_header H9ET92WW.CAP.tmp)
mv H9ET92WW.CAP.tmp H9ET92WW.CAP
Downloading l430,l530 BIOS 2.54 (G3ET94WW) EC 1.14 (G3HT40WW)
2022-12-01 09:15:03 URL:https://download.lenovo.com/pccbbs/mobiles/g3uj13us.iso [25538560/25538560] -> "g3uj13us.iso.orig" [1]
scripts/checksum --mv_on_fail g3uj13us.iso.orig g3uj13us.iso
Downloading b590,b490 BIOS ?.? (H9ET75WW) EC 1.01 (H9EC08WW)
2022-12-01 09:15:15 URL:https://download.lenovo.com/pccbbs/mobiles/h9et75ww.zip [49576764/49576764] -> "h9et75ww.zip.orig" [1]
scripts/checksum --mv_on_fail h9et75ww.zip.orig h9et75ww.zip
mkdir -p h9et92ww.exe.orig.extract.tmp
unzip -o h9et75ww.zip.orig DOS/\* -x \*.cap \*.IMC \*.BAT -d h9et92ww.exe.orig.extract.tmp/
Archive:  h9et75ww.zip.orig
   creating: h9et92ww.exe.orig.extract.tmp/DOS/
  inflating: h9et92ww.exe.orig.extract.tmp/DOS/Bios.EXE  
  inflating: h9et92ww.exe.orig.extract.tmp/DOS/chkGPIO.EXE  
  inflating: h9et92ww.exe.orig.extract.tmp/DOS/CMPBIOS.EXE  
  inflating: h9et92ww.exe.orig.extract.tmp/DOS/DosFlash.exe  
  inflating: h9et92ww.exe.orig.extract.tmp/DOS/EC.EXE  
  inflating: h9et92ww.exe.orig.extract.tmp/DOS/ECIO.EXE  
  inflating: h9et92ww.exe.orig.extract.tmp/DOS/EEP_INIT.COM  
 extracting: h9et92ww.exe.orig.extract.tmp/DOS/ERROR.LOG  
  inflating: h9et92ww.exe.orig.extract.tmp/DOS/FIND.EXE  
  inflating: h9et92ww.exe.orig.extract.tmp/DOS/fparts.txt  
  inflating: h9et92ww.exe.orig.extract.tmp/DOS/fpt.exe  
  inflating: h9et92ww.exe.orig.extract.tmp/DOS/GPIO.EXE  
 extracting: h9et92ww.exe.orig.extract.tmp/DOS/GPIO.INI  
  inflating: h9et92ww.exe.orig.extract.tmp/DOS/SMBIOS.EXE  
mattrib -i h9et92ww.exe.tmp@@71680 -r -/ ::FLASH/
mdeltree -i h9et92ww.exe.tmp@@71680 FLASH/
mmd -i h9et92ww.exe.tmp@@71680 FLASH
mcopy -o -s -m -i h9et92ww.exe.tmp@@71680 h9et92ww.exe.orig.extract.tmp/DOS/* ::/FLASH/
mcopy -t -m -o -i h9et92ww.exe.tmp@@71680 H9ET92WW.CAP.tmp ::/FLASH/H9ET92WW.CAP
mcopy -t -m -o -i h9et92ww.exe.tmp@@71680 h9et92ww.exe.report.tmp ::report.txt
mcopy -t -m -o -i h9et92ww.exe.tmp@@71680 h9et92ww.exe.bat.tmp ::AUTOEXEC.BAT
file is read only, overwrite anyway (y/n) ? y
cp h9et92ww.exe patched.b590.iso
cp h9et92ww.exe.report patched.b590.iso.report

Your build has completed with the following details:

Built ISO: 9182a061ae2f238eaebf5130a45dfcbdca9229f8  patched.b590.iso
Based on code from: b590,b490 BIOS ?.? (H9ET92WW) EC 1.02 (H9EC09WW)
Buildinfo:  (20221201) patched.b590.img
Built FL2: 1647da4d2edb03e61a764cbe9614f506113814f0  H9ET92WW.CAP

Patches applied:
b590.H9ET92WW.img.d/006_battery_validate.patch

./scripts/geteltorito -o patched.b590.img.tmp patched.b590.iso
Booting catalog starts at sector: 20 
Manufacturer of CD: NERO BURNING ROM
Image architecture: x86
Boot media type is: harddisk
El Torito image starts at sector 27 and has 49152 sector(s) of 512 Bytes

Image has been written to file "patched.b590.img.tmp".
./scripts/fix_mbr patched.b590.img.tmp

Your build has completed with the following details:

Built ISO: 9182a061ae2f238eaebf5130a45dfcbdca9229f8  patched.b590.iso
Based on code from: b590,b490 BIOS ?.? (H9ET92WW) EC 1.02 (H9EC09WW)
Buildinfo:  (20221201) patched.b590.img
Built FL2: 1647da4d2edb03e61a764cbe9614f506113814f0  H9ET92WW.CAP

Patches applied:
b590.H9ET92WW.img.d/006_battery_validate.patch
`

deeptenk2, 2022-12-01:
- author@ I checked, the result is the same. Both under Windows and under dos. The file sizes are also different.I still ask you to make me a working * .img and * .bin for the programmer. Thank you in ***@***.***:~/thinkpad-ec-Lx30$ make patched.b590.imgGenerated dependancies from descriptionsDownloading b590,b490 BIOS ?.? (H9ET92WW) EC 1.02 (H9EC09WW)2022-12-01 09:14:54 URL:https://download.lenovo.com/pccbbs/mobiles/h9et92ww.exe [4525327/4525327] -> "h9et92ww.exe.orig" [1]scripts/checksum --mv_on_fail h9et92ww.exe.orig h9et92ww.exemv `innoextract h9et92ww.exe.orig -I H9ET92WW.CAP | grep -i H9ET92WW.CAP | cut -d'"' -f2` H9ET92WW.CAP.orig./scripts/FL2_copyIMG from_fl2 H9ET92WW.CAP.orig b590.H9ET92WW.img.origIMG at offset 0x4001d0 size 0x20000 (EFI::Capsule::PFH_header H9ET92WW.CAP.orig)./scripts/hexpatch.pl --rm_on_fail --fail_on_missing --report b590.H9ET92WW.img.report b590.H9ET92WW.img b590.H9ET92WW.img.d/006_battery_validate.patchAttempting to patch b590.H9ET92WW.imgApplying b590.H9ET92WW.img.d/006_battery_validate.patch -4178,14 +4178,14mkdir nuvoton-toolswget -O nuvoton-tools/npce885crc.c https://raw.githubusercontent.com/leecher1337/thinkpad-Lx30-ec/main/fwpat/util/npce885crc.c--2022-12-01 09:14:55--  https://raw.githubusercontent.com/leecher1337/thinkpad-Lx30-ec/main/fwpat/util/npce885crc.cРаспознаётся raw.githubusercontent.com (raw.githubusercontent.com)… 185.199.109.133, 185.199.110.133, 185.199.111.133, ...Подключение к raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... соединение установлено.HTTP-запрос отправлен. Ожидание ответа… 200 OKДлина: 3276 (3,2K) [text/plain]Сохранение в: ‘nuvoton-tools/npce885crc.c’ nuvoton-tools/npce8 100%[===================>]   3,20K  --.-KB/s    за 0,1s     2022-12-01 09:14:56 (27,6 KB/s) - ‘nuvoton-tools/npce885crc.c’ сохранён [3276/3276] gcc -o nuvoton-tools/npce885crc nuvoton-tools/npce885crc.cfatal: не найден git репозиторий (или один из родительских каталогов): .git./nuvoton-tools/npce885crc -o 0x8000 -u b590.H9ET92WW.imgNUVOTON Insyde firmware CRC calculator v1.00(c) ***@***.*** 02/2022 Searching at offset 8000CRC Start address=0048800AAssumed load address: 00488000CRC indexing offset = ACalculated CRC = A2, file CRC = 75 --> CRC differs!Updating CRC...OK!cp --reflink=auto H9ET92WW.CAP.orig H9ET92WW.CAP.tmp./scripts/FL2_copyIMG to_fl2 H9ET92WW.CAP.tmp b590.H9ET92WW.imgIMG at offset 0x4001d0 size 0x20000 (EFI::Capsule::PFH_header H9ET92WW.CAP.tmp)mv H9ET92WW.CAP.tmp H9ET92WW.CAPDownloading l430,l530 BIOS 2.54 (G3ET94WW) EC 1.14 (G3HT40WW)2022-12-01 09:15:03 URL:https://download.lenovo.com/pccbbs/mobiles/g3uj13us.iso [25538560/25538560] -> "g3uj13us.iso.orig" [1]scripts/checksum --mv_on_fail g3uj13us.iso.orig g3uj13us.isoDownloading b590,b490 BIOS ?.? (H9ET75WW) EC 1.01 (H9EC08WW)2022-12-01 09:15:15 URL:https://download.lenovo.com/pccbbs/mobiles/h9et75ww.zip [49576764/49576764] -> "h9et75ww.zip.orig" [1]scripts/checksum --mv_on_fail h9et75ww.zip.orig h9et75ww.zipmkdir -p h9et92ww.exe.orig.extract.tmpunzip -o h9et75ww.zip.orig DOS/\* -x \*.cap \*.IMC \*.BAT -d h9et92ww.exe.orig.extract.tmp/Archive:  h9et75ww.zip.orig   creating: h9et92ww.exe.orig.extract.tmp/DOS/  inflating: h9et92ww.exe.orig.extract.tmp/DOS/Bios.EXE    inflating: h9et92ww.exe.orig.extract.tmp/DOS/chkGPIO.EXE    inflating: h9et92ww.exe.orig.extract.tmp/DOS/CMPBIOS.EXE    inflating: h9et92ww.exe.orig.extract.tmp/DOS/DosFlash.exe    inflating: h9et92ww.exe.orig.extract.tmp/DOS/EC.EXE    inflating: h9et92ww.exe.orig.extract.tmp/DOS/ECIO.EXE    inflating: h9et92ww.exe.orig.extract.tmp/DOS/EEP_INIT.COM   extracting: h9et92ww.exe.orig.extract.tmp/DOS/ERROR.LOG    inflating: h9et92ww.exe.orig.extract.tmp/DOS/FIND.EXE    inflating: h9et92ww.exe.orig.extract.tmp/DOS/fparts.txt    inflating: h9et92ww.exe.orig.extract.tmp/DOS/fpt.exe    inflating: h9et92ww.exe.orig.extract.tmp/DOS/GPIO.EXE   extracting: h9et92ww.exe.orig.extract.tmp/DOS/GPIO.INI    inflating: h9et92ww.exe.orig.extract.tmp/DOS/SMBIOS.EXE  mattrib -i h9et92ww.exe.tmp@@71680 -r -/ ::FLASH/mdeltree -i h9et92ww.exe.tmp@@71680 FLASH/mmd -i h9et92ww.exe.tmp@@71680 FLASHmcopy -o -s -m -i h9et92ww.exe.tmp@@71680 h9et92ww.exe.orig.extract.tmp/DOS/* ::/FLASH/mcopy -t -m -o -i h9et92ww.exe.tmp@@71680 H9ET92WW.CAP.tmp ::/FLASH/H9ET92WW.CAPmcopy -t -m -o -i h9et92ww.exe.tmp@@71680 h9et92ww.exe.report.tmp ::report.txtmcopy -t -m -o -i h9et92ww.exe.tmp@@71680 h9et92ww.exe.bat.tmp ::AUTOEXEC.BATfile is read only, overwrite anyway (y/n) ? ycp h9et92ww.exe patched.b590.isocp h9et92ww.exe.report patched.b590.iso.report Your build has completed with the following details: Built ISO: 9182a061ae2f238eaebf5130a45dfcbdca9229f8  patched.b590.isoBased on code from: b590,b490 BIOS ?.? (H9ET92WW) EC 1.02 (H9EC09WW)Buildinfo:  (20221201) patched.b590.imgBuilt FL2: 1647da4d2edb03e61a764cbe9614f506113814f0  H9ET92WW.CAP Patches applied:b590.H9ET92WW.img.d/006_battery_validate.patch ./scripts/geteltorito -o patched.b590.img.tmp patched.b590.isoBooting catalog starts at sector: 20 Manufacturer of CD: NERO BURNING ROMImage architecture: x86Boot media type is: harddiskEl Torito image starts at sector 27 and has 49152 sector(s) of 512 Bytes Image has been written to file "patched.b590.img.tmp"../scripts/fix_mbr patched.b590.img.tmp Your build has completed with the following details: Built ISO: 9182a061ae2f238eaebf5130a45dfcbdca9229f8  patched.b590.isoBased on code from: b590,b490 BIOS ?.? (H9ET92WW) EC 1.02 (H9EC09WW)Buildinfo:  (20221201) patched.b590.imgBuilt FL2: 1647da4d2edb03e61a764cbe9614f506113814f0  H9ET92WW.CAP Patches applied:b590.H9ET92WW.img.d/006_battery_validate.patch`

Then you should receive a correct .iso file that you can boot and patch. If you are still having problems, I can also patch your BIOS dump, but I don't think it will be necessary.

deeptenk2, 2022-12-04:
Guys, where are you?

deeptenk2, 2022-12-08:
Thanks, I'll try the dump tomorrow. For me, flashing through a real programmer is the easiest solution. I will inform you about the result. My friend and I have been waiting for your reply.

deeptenk2, 2022-12-08:
> You can simply patch your dump with;./patchui.sh patchdump B590 "bat" ~/b590.bin

Thanks, I'll try the dump tomorrow. For me, flashing through a real programmer is the easiest solution. I will inform you about the result. My friend and I have been waiting for your reply.

deeptenk2, 2022-12-09:
I flashed http://dose.0wnz.at/b590.zip with a programmer. Everything is great! Huge thanks and respect!

closed 2022-12-09 by deeptenk2

12thDimension, 2025-05-06:
I actually have the same issue. I ordered a programmer now but didnt really understand what my next steps would be. Im new to flashing bios/ec

leecher1337, 2025-05-07:
There was a Typo in the generation script, so you can use my fork to build:

```
git clone https://github.com/leecher1337/thinkpad-ec.git
cd thinkpad-ec/
make patch_disable_keyboard clean
make patch_enable_battery clean
make patched.b590.img
```

If you want to do manual flashing anyway:

> I actually have the same issue. I ordered a programmer now but didnt really understand what my next steps would be. Im new to flashing bios/ec

I made a short description here:
https://github.com/leecher1337/thinkpad-Lx30-ec?tab=readme-ov-file#external-flashing-with-a-programmer

Raspi works best for flashing, at least for me. If you bought a CH341A < Version 1.6, don't forget to do the 3.3v fix to not brick your machine. Version 1.6 or 1.7 has a voltage selector switch and may be OK. But I never had success with these devices.

12thDimension, 2025-05-09:
Thanks for the reply. I bought this one: https://www.amazon.de/dp/B09MHH6D8H?ref=ppx_yo2ov_dt_b_fed_asin_title
but couldnt get it to work. The laptop still boots so I didnt accidentally fry the chip.
Im thinking of desoldering the chip for a better connection to the programmer. That has to wait for at least a week though. 

Ill try your fork in the meantime and will write an update

12thDimension, 2025-05-09:
Okay I get a new error code (14). I believe the other guy got that too?
Meaning I probably have to wait for the desoldering. Do you have any ideas what else I could try?

leecher1337, 2025-05-10:
Ah, yes, there was this error and as I don't have such a model, I was unable to find out about it.

I found out that some of the Notebook models need the `/sn` dosflash parameter to continue, As you have an external flasher anyway for potential recovery and are willing to desolder the chip, you may want to try it out (do it at your own risk).
The easiest way possibly is to modify the autoexec.bat.template file before building your image and adding `/sn` after `/sd` in the file, then rebuild the image and then try it again.
Let me know if this worked. If so, I'd make another pull request to include this parameter.

So you bought the black CHA431A that usually suffers from the 5V level bug. Yeah, in most cases it goes well and it is not likely that you will fry your chip with it (neglegible current when at 5V in idle mode and when you write, it goes down to correct 3.3V level), but you can read a bit about the problem here:
https://libreboot.org/docs/install/spi.html#do-not-buy-ch341a

Having said that, I also never got this CHA341A junk to work properly, I was about to give up with it on the L430 when I then gave my Raspberry Pi a try and with it, it worked like a charm.
On said page, you also have the Raspebrry Pi Pico which is far cheaper than the original Raspi that I used, so if you are uncomfortable desoldering the chip, you may want to try it with Raspi pico (I assume that it works like with the normal Raspi that I used).
Maybe @deeptenk2 can tell you which programmer he successfully used.

deeptenk2, 2025-05-11:
> Okay I get a new error code (14). I believe the other guy got that too?
> Meaning I probably have to wait for the desoldering. Do you have any ideas what else I could try?

> Maybe [@deeptenk2](https://github.com/deeptenk2) can tell you which programmer he successfully used.

I used ch341a black + soldering chip + neoprogrammer software, now i am more experienced and i know many ways to cheat a laptop without a programmer, but i need to know the details. you can contact me in telegram @deeptenk

12thDimension, 2025-05-22:
Update for @leecher1337 and anyone else stumbling upon this thread:
I didnt manage to desolder the chip and frankly Im thinking of abandoning this project. Maybe I'll give the raspy way a try

leecher1337, 2025-05-22:
Did you try the /sn switch? Maybe there is no need to desolder the chip

12thDimension, 2025-05-26:
I just tried it but sadly same result. "ERROR 081 - Failed to load BIOS image file! Status = 14."
Is there any log I can get that would help pinning it down?

leecher1337, 2025-09-20:
Unfortunately the only way to find out what this is is to debug it on a live machine (under DOS, which doesn't make it easier, yay), as there is no documentation for the error codes.
I found a B590 machine for 50 EUR (incl. shipping) today, if someone is willing to cover the costs of the machine, I can buy it and debug the flasher and find a way how to flash the BIOS correctly to this machine.
https://www.willhaben.at/iad/kaufen-und-verkaufen/d/lenovo-b590-6274-i3-3110m-6gb-ram-480gb-ssd-1337199920/
But I guess we would need to be fast, normally it doesn't take long until someone buys it.

leecher1337, 2025-09-21:
Maybe the solution is really simple. It must be noted that h9et92ww doesn't ship with dosflash in an ISO image, that's why on building we're "stealing" dosflash from another iso and just place the capsule in there.
It may be possible that dosflash simply is incompatible with the capsule format and cannot identify the ec region in the ec capsule correctly.
So we could try to use the original winflash64 under windows that normally gets used by the pc vendor to do the bios update.
Find attached a package with flash.bat that uses winflash to flash the modified capsule instead of dosflash on a bootable iso.
Use at your own risk! As mentioned, I don't have such a machine, so I can't try it out myself.

[flashpkg.zip](https://github.com/user-attachments/files/22451595/flashpkg.zip)

If it works, I can modify the build scripts to create such a windows package instead of dosflash iso.

12thDimension, 2025-09-30:
Thanks a lot for your effort to help me. Im a bit scared now to use the zip as I have no way of verifying its content

leecher1337, 2025-12-14:
It also seems that the B590 has different Firmware and BIOS versions: H1ET85WW, H5ET85WW, H9ET92WW
And if you check [h9et92ww.zip](https://download.lenovo.com/pccbbs/mobiles/h9et92ww.zip) there seem to be different detection methods for different BIOS versions and there are also .imc files depending on SMBIOS variables, so you may want to check the detection methods shown there in order to find out if H9ET92WW.cap is suitable on your machine.

I now added support for these other firmware versions aswell:
https://github.com/hamishcoleman/thinkpad-ec/pull/278