hamishcoleman/thinkpad-ec #232: Heads-up: X230 BIOS 2.75 (G2ETB5WW) not bootable from USB

clickworkorange, 2023-01-21:
I recently got yet _another_ X230, which had already been updated to BIOS 2.77 (ECP 1.15), and of course I wanted to swap in a proper keyboard, and I needed to downgrade the BIOS to <=2.75 in order to apply the thinkpad-ec patches. I already had not one but _two_ downloaded copies of the 2.75 BIOS, but could not get either of these to boot from a USB stick: 

    $ geteltorito -o bios.img BIOS\ v2.75\,\ ECP\ v1.14\ g2uj31us.iso
    $ sudo dd if=bios.img of=/dev/sdb bs=1M

The stick was detected as a bootable USB hard drive, but trying to boot from it just sent me back to the boot drive selection menu. Contents looked ok in Thunar and GParted. Although the same USB stick had booted fine with the thinkpad-ec patcher iso I first suspected some issue with the USB media, so I tried swapping it for another stick, but still no luck. I also tried copying the .iso straight to the stick without the El Torito processing. Finally I tried downloading the .iso again from Lenovo, but I still could not get it to work. As soon as I tried a different BIOS file (2.68) it booted without a hitch - though the flashing software claimed that "the system program is not correct for this system". I then grabbed 2.74 and this booted and allowed the downgrade. So it seems to me the X230 2.75 BIOS iso has some issues - has anyone else noticed this?

ltvkit, 2023-04-13:
Currently having some issues. But your issue sounds like you're trying to boot in UEFI mode, not Legacy.

diabl0w, 2023-07-21:
im also having this issue, I can only guess that lenovo changed the file on their server? because this github specifically committed to add this bios version so I would think it would have been tested first

BachoSeven, 2023-08-28:
@hamishcoleman any news on this? In case, is it possible to patch 2.74 or an older version? I have yet to try this, as soon as I mod the x220 keyboard for my x230 I will let you guys know how it goes

hamishcoleman, 2023-08-28:
The downloader has known good checksums of the ISO files, so it Lenovo had changed anything, it would show up on the automated CI.

It does appear to be simple to get the BIOS confused about what you are trying to boot, but that just requires that you try to match the correct boot settings.

There are no known reasons why any of this patcher has stopped functioning - so you should be good to just use it.

BachoSeven, 2023-08-30:
Great to hear, thanks for the response!

diabl0w, 2023-08-30:
I would like to retract my "metoo" on this issue to say that it was user error. I cant remember what exactly the issue was, but if i had to guess, i think i was trying to boot from Ventoy instead of writing the image directly to usb.