hamishcoleman/thinkpad-ec #234: Can't rollback to T430 layout

tmmyplffe, 2023-02-11:
I've tried to rollback to the T430 keyboard layout after my T420 keyboard stopped working.

I entered the following lines in the terminal, and yet, the layout won't change back to the original one:

sudo apt-get update
sudo apt-get install make git
cd ~/
git clone https://github.com/hamishcoleman/thinkpad-ec
cd ~/thinkpad-ec
sudo make build-deps
make list_laptops
**sudo make patch_disable_keyboard clean**
sudo make patched.x230.img
sudo dd if=patched.x230.img of=/dev/sdx bs=4M status=progress conv=fsync

Have I misunderstood the COFIG document? Am I doing something wrong here?

hamishcoleman, 2023-02-11:
That all looks like you are doing it right.  Can you post the full output of your build?

tmmyplffe, 2023-02-12:
user@OEMCOMPUTER:~$ **_cd ~/Downloads/thinkpad-ec_**
user@OEMCOMPUTER:~/Downloads/thinkpad-ec$ **_sudo make build-deps_**
[sudo] password for user: 
apt -y install \
    git \
    mtools \
    libssl-dev \
    build-essential \
    xorriso \
    unzip \
    innoextract \

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
build-essential is already the newest version (12.9ubuntu3).
mtools is already the newest version (4.0.33-1+really4.0.32-1build1).
innoextract is already the newest version (1.8-1.2build1).
xorriso is already the newest version (1.5.4-2).
git is already the newest version (1:2.34.1-1ubuntu1.6).
libssl-dev is already the newest version (3.0.2-0ubuntu1.8).
unzip is already the newest version (6.0-26ubuntu3.1).
0 upgraded, 0 newly installed, 0 to remove and 344 not upgraded.
user@OEMCOMPUTER:~/Downloads/thinkpad-ec$ **_sudo make list_laptops_**

The following make targets are the supported usb images:

patched.t430.img	- for patching Thinkpad T430
patched.t430s.img	- for patching Thinkpad T430s
patched.t530.img	- for patching Thinkpad T530
patched.t530i.img	- for patching Thinkpad T530i
patched.w530.img	- for patching Thinkpad W530
patched.x230.img	- for patching Thinkpad X230
patched.x230t.img	- for patching Thinkpad X230t

user@OEMCOMPUTER:~/Downloads/thinkpad-ec$ **_sudo make patch_disable_keyboard clean_**
sed -E 's/CONFIG_KEYBOARD\s*=.*//' --in-place .config
echo "CONFIG_KEYBOARD = n" >> .config
rm -f .d/generated.deps \
            patched.*.iso patched.*.img *.FL? *.FL?.orig *.img.enc \
            *.img.enc.orig *.img.orig *.bat *.report \
            *.img \
            *.txt.orig
rm -rf *.iso.extract *.iso.orig.extract
user@OEMCOMPUTER:~/Downloads/thinkpad-ec$ sudo make patched.t430.img
Generated dependancies from descriptions
Downloading t430 BIOS 2.81 (G1ETC1WW) EC 1.13 (G1HT35WW)
2023-02-12 12:11:08 URL:https://download.lenovo.com/pccbbs/mobiles/g1uj48us.iso [39170048/39170048] -> "g1uj48us.iso.orig" [1]
scripts/checksum --mv_on_fail g1uj48us.iso.orig g1uj48us.iso
./scripts/ISO_copyFL2 from_iso g1uj48us.iso.orig t430.G1HT35WW.s01D2000.FL2.orig 01D2000.FL2
git submodule update --init --remote
Submodule 'mec-tools' (https://github.com/eigenmatt/mec-tools) registered for path 'mec-tools'
Cloning into '/home/user/Downloads/thinkpad-ec/mec-tools'...
Submodule path 'mec-tools': checked out '07a1b1407326488376afed6c596d37c53d7432b6'
git submodule update
make -C mec-tools
make[1]: Entering directory '/home/user/Downloads/thinkpad-ec/mec-tools'
cc -Wall -O2 mec_csum_boot.c -o mec_csum_boot 
cc -Wall -O2 mec_csum_flasher.c -o mec_csum_flasher 
cc -Wall -O2 mec_csum_outer.c -o mec_csum_outer 
cc -Wall -O2 mec_encrypt.c -o mec_encrypt -lcrypto
mec_encrypt.c: In function ‘main’:
mec_encrypt.c:184:5: warning: ‘BF_cbc_encrypt’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  184 |     BF_cbc_encrypt(base, out, 0x200, &key, thisiv, enc);
      |     ^~~~~~~~~~~~~~
In file included from mec_encrypt.c:6:
/usr/include/openssl/blowfish.h:57:28: note: declared here
   57 | OSSL_DEPRECATEDIN_3_0 void BF_cbc_encrypt(const unsigned char *in,
      |                            ^~~~~~~~~~~~~~
mec_encrypt.c:188:5: warning: ‘BF_cbc_encrypt’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
[... 68 lines cut ...]
Built FL2: 617a85d1a4df92ef1171369d6c8a168b8e864e87  t430.G1HT35WW.s01D2000.FL2

No patches applied: Lenovo default firmware

user@OEMCOMPUTER:~/Downloads/thinkpad-ec$ **_lsblk -d -o NAME,SIZE,LABEL_**
NAME     SIZE LABEL
loop1   45.9M 
loop2  248.8M 
loop3   81.3M 
loop4   61.9M 
loop5  155.6M 
loop6      4K 
loop7    284K 
loop8   49.8M 
loop9   63.3M 
loop10  91.7M 
loop11   304K 
loop12  45.9M 
loop13 346.3M 
sda    111.8G 
sdb    465.8G 
user@OEMCOMPUTER:~/Downloads/thinkpad-ec$ **_sudo dd if=patched.t430.img of=/dev/sdc bs=4M status=progress conv=fsync_**
9+1 records in
9+1 records out
38797312 bytes (39 MB, 37 MiB) copied, 3.8496 s, 10.1 MB/s

tmmyplffe, 2023-02-12:
user@OEMCOMPUTER:~$ cd ~/Downloads/thinkpad-ec
user@OEMCOMPUTER:~/Downloads/thinkpad-ec$ sudo make build-deps
[sudo] password for user: 
apt -y install \
    git \
    mtools \
    libssl-dev \
    build-essential \
    xorriso \
    unzip \
    innoextract \

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
build-essential is already the newest version (12.9ubuntu3).
mtools is already the newest version (4.0.33-1+really4.0.32-1build1).
innoextract is already the newest version (1.8-1.2build1).
xorriso is already the newest version (1.5.4-2).
git is already the newest version (1:2.34.1-1ubuntu1.6).
libssl-dev is already the newest version (3.0.2-0ubuntu1.8).
unzip is already the newest version (6.0-26ubuntu3.1).
0 upgraded, 0 newly installed, 0 to remove and 344 not upgraded.
user@OEMCOMPUTER:~/Downloads/thinkpad-ec$ sudo make list_laptops

The following make targets are the supported usb images:

patched.t430.img	- for patching Thinkpad T430
patched.t430s.img	- for patching Thinkpad T430s
patched.t530.img	- for patching Thinkpad T530
patched.t530i.img	- for patching Thinkpad T530i
patched.w530.img	- for patching Thinkpad W530
patched.x230.img	- for patching Thinkpad X230
patched.x230t.img	- for patching Thinkpad X230t

user@OEMCOMPUTER:~/Downloads/thinkpad-ec$ sudo make patch_disable_keyboard clean
sed -E 's/CONFIG_KEYBOARD\s*=.*//' --in-place .config
echo "CONFIG_KEYBOARD = n" >> .config
rm -f .d/generated.deps \
            patched.*.iso patched.*.img *.FL? *.FL?.orig *.img.enc \
            *.img.enc.orig *.img.orig *.bat *.report \
            *.img \
            *.txt.orig
rm -rf *.iso.extract *.iso.orig.extract
user@OEMCOMPUTER:~/Downloads/thinkpad-ec$ sudo make patched.t430.img
Generated dependancies from descriptions
Downloading t430 BIOS 2.81 (G1ETC1WW) EC 1.13 (G1HT35WW)
2023-02-12 12:11:08 URL:https://download.lenovo.com/pccbbs/mobiles/g1uj48us.iso [39170048/39170048] -> "g1uj48us.iso.orig" [1]
scripts/checksum --mv_on_fail g1uj48us.iso.orig g1uj48us.iso
./scripts/ISO_copyFL2 from_iso g1uj48us.iso.orig t430.G1HT35WW.s01D2000.FL2.orig 01D2000.FL2
git submodule update --init --remote
Submodule 'mec-tools' (https://github.com/eigenmatt/mec-tools) registered for path 'mec-tools'
Cloning into '/home/user/Downloads/thinkpad-ec/mec-tools'...
Submodule path 'mec-tools': checked out '07a1b1407326488376afed6c596d37c53d7432b6'
git submodule update
make -C mec-tools
make[1]: Entering directory '/home/user/Downloads/thinkpad-ec/mec-tools'
cc -Wall -O2 mec_csum_boot.c -o mec_csum_boot 
cc -Wall -O2 mec_csum_flasher.c -o mec_csum_flasher 
cc -Wall -O2 mec_csum_outer.c -o mec_csum_outer 
cc -Wall -O2 mec_encrypt.c -o mec_encrypt -lcrypto
mec_encrypt.c: In function ‘main’:
mec_encrypt.c:184:5: warning: ‘BF_cbc_encrypt’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  184 |     BF_cbc_encrypt(base, out, 0x200, &key, thisiv, enc);
      |     ^~~~~~~~~~~~~~
In file included from mec_encrypt.c:6:
/usr/include/openssl/blowfish.h:57:28: note: declared here
   57 | OSSL_DEPRECATEDIN_3_0 void BF_cbc_encrypt(const unsigned char *in,
      |                            ^~~~~~~~~~~~~~
mec_encrypt.c:188:5: warning: ‘BF_cbc_encrypt’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
[... 68 lines cut ...]
Built FL2: 617a85d1a4df92ef1171369d6c8a168b8e864e87  t430.G1HT35WW.s01D2000.FL2

No patches applied: Lenovo default firmware

user@OEMCOMPUTER:~/Downloads/thinkpad-ec$ lsblk -d -o NAME,SIZE,LABEL
NAME     SIZE LABEL
loop1   45.9M 
loop2  248.8M 
loop3   81.3M 
loop4   61.9M 
loop5  155.6M 
loop6      4K 
loop7    284K 
loop8   49.8M 
loop9   63.3M 
loop10  91.7M 
loop11   304K 
loop12  45.9M 
loop13 346.3M 
sda    111.8G 
sdb    465.8G 
user@OEMCOMPUTER:~/Downloads/thinkpad-ec$ sudo dd if=patched.t430.img of=/dev/sdc bs=4M status=progress conv=fsync
9+1 records in
9+1 records out
38797312 bytes (39 MB, 37 MiB) copied, 3.8496 s, 10.1 MB/s

hamishcoleman, 2023-02-12:
Well, your `Built FL2:` hash matches the one I got, so you should have a good image to flash.

Do you see any BIOS messages about firmware updating (and a percentage complete bar) when you next power on the computer?  If you never see those, then the image didnt make it to the EC chip

tmmyplffe, 2023-02-12:
I boot the USB key by pressing F12. Then this is what I got (copied from pictures, I might miss a few lines of text):

This will flash your Embedded controller firmware

WARNING: Incorrectly flashing your firmware can damage your laptop.
Ensure this is the right firmware version before attempting to flash
the firmware - there are minimal checks done.

This disc will flash the file G1ETC1WW\$01D2000.FL2

Based on code from: t430 BIOS 2.81 (G1ETC1WW) EC 1.13 (G1HT35WW) 
Buildinfo: v1-431-g75383a (20230208) patched.t430.img
Built FL2: 617a85d1a4df92ef1171369d6c8a168b8e864e87 t430.G1HT35WW.s01D2000.FL2

No patches applied: Lenovo default firmware

Press any key to continue . . .

Copyright (c) 2011-2012 Phoenix Technologies Ltd.
Copyright (c) 2011-2012 Lenovo Group Limited.

Read BIOS image from file.
Initialize Flash module.
Read current BIOS.
Oem check

Prepare to flash "ec"

Do not turn off the computer during the update!!!

Begin Flashing......
Total blocks of the image = 48.
:---+----+----+----+----+----+----+----+----+----:
..................................................
Image flashing done.

Flashing finished.

BIOS is updated successfully.

WARNING: System will shutdown or reboot in 5 seconds!

hamishcoleman, 2023-02-12:
What it doesnt say from the flash tool is that this is only phase-1 -
after the system shuts down, the bootup BIOS screen also needs to go through a flash sequence.

You should see another set of messages at bootup and when that happens.  We believe that the BIOS checks for a couple of other things before letting this happen, but it is all guesswork - we think that it looks for a >80% charged battery and a charger plugged in (sometimes it seems like it needs either, some reports were both, some people reported that removing the battery made this work - nothing is clear)

tmmyplffe, 2023-02-13:
I finally got it working. Thank you so much for your help. I couldn't have done it without you!

Although I had a downgraded BIOS for previous attempts, I re-installed the 1vyrain BIOS given that I couldn't flash the embedded controller (I also prioritized being able to use my blacklisted WiFi card).
Although, unlike previous attempts, I used an OEM battery this time and it worked like a charm.

Here is what I had to do:

1 - Downgrade the 1vyrain BIOS to 2.64 and remove the blacklisted WiFi card
2 - Use an OEM battery and leave the charger plugged in
3 - Flash the embedded controller using the patched.t430.img and let the BIOS do it's thing on reboot
4 - Flash the 1vyrain BIOS
5 - re-install the WiFi card