hamishcoleman/thinkpad-ec #238: b590 keysym.patch

kuraimangetsu, 2023-06-01:
hi
trying to patch b590 battery issue and i get "Cannot read b590.H9ET92WW.img.d/001_keysym.patch"
and there is no such file either in the folder!
so what to do?

kuraimangetsu, 2023-06-02:
found it on leecher1337/thinkpad-ec)
made the image and now am getting
error 081 - faild to load bios image file! status =14

the bios image is H9ET92WW
and the bios flashed with newer one
can't roll back
cuz the system says it's not matching the machine and halted

hamishcoleman, 2023-06-03:
I am not aware of any keyboard patch available for the b590,which is why you were getting that error - if you just wanted the battery patch, you simply need to disable the keyboard patch.

I'm not sure if I understand what you are reporting in your second comment.  It sounds like you might have had issues with the Lenovo flasher tool or the laptop itself - unfortunately, these are poorly understood and if you flash an incorrect image there is not a lot I know how to do.