hamishcoleman/thinkpad-ec #240: git clone https://github.com/hamishcoleman/thinkpad-ec doesn't work

lenovo878, 2023-06-17:
when I pasted git clone https://github.com/hamishcoleman/thinkpad-ec in it says 
Cloning into 'thinkpad-ec'...
"fatal: unable to access" https://github.com/hamishcoleman/thinkpad-ec/': server certificate verification failed. CAfile: /etc/ssl/cert/ca-certificates.crt CRLfile: none

Any ideas?

hamishcoleman, 2023-06-20:
This sounds like your system is out of date.  What distribution are you running your clone on?

("server certificate verification failed" is usually due to out of date root certificates on the client end)

lenovo878, 2023-06-21:
Sorry, but are you saying I need an updated version of Linux?

hamishcoleman, 2023-06-21:
You might need an updated version of Linux, yes.  What version are you using?

lenovo878, 2023-06-21:
Linux mint version 14.04

hamishcoleman, 2023-06-21:
That distribution is over ten years old, so the SSL certificates it knows about have all changed (and often expired).  I am surprised you have not had the same issues with accessing anything else on the internet

lenovo878, 2023-06-21:
Thanks, upgraded to the latest version now, which is 21.1.
However, now stuck on the below.
After I do cd thinkpad-ec
I try and type in make patch_enable_battery clean and I get the below.

oot@mint:/home/mint/thinkpad-ec# make patch_enable_battery clean
Generated dependancies from descriptions
sed -E 's/CONFIG_BATTERY\s*=.*//' --in-place .config
echo "CONFIG_BATTERY = y" >> .config
rm -f .d/generated.deps \
            patched.*.iso patched.*.img *.FL? *.FL?.orig *.img.enc \
            *.img.enc.orig *.img.orig *.bat *.report \
            *.img \
            *.txt.orig
rm -rf *.iso.extract *.iso.orig.extract

hamishcoleman, 2023-06-21:
I'm not seeing any errors or issues in the output you have there - have you tried simply continuing on to the next step?

lenovo878, 2023-06-21:
Tried it again and it works now. Thanks for all your help!