hamishcoleman/thinkpad-ec #246: x230: Trying to boot from USB does nothing, returns to menu

dcht00, 2023-09-28:
It just blinks for a split second, and returns straight to the boot options menu.
Tried different USB keys, ports ... No luck.
Fully charged original battery.

My current bios is G2ETB3WW (2.73), EC 1.14.
Running "legacy first" mode, ...

--------------------- iso report
Based on code from: x230 BIOS 2.75 (G2ETB5WW) EC 1.14 (G2HT35WW)
Buildinfo: v1-435-g1047da (20230927) patched.x230.iso
Built FL2: 611c88f62add31aa4cdbedeec50e9583af509cbe  x230.G2HT35WW.s01D3000.FL2

Patches applied:
x230.G2HT35WW.img.d/006_battery_validate.patch

--------------------- generating USB
$ sudo dd if=patched.x230.iso of=/dev/sdc bs=4M status=progress conv=fsync
8+1 records in
8+1 records out
33927168 bytes (34 MB, 32 MiB) copied, 1.71633 s, 19.8 MB/s

Taiphee98, 2025-08-22:
Were you able to resolve this?

msoczewka, 2025-08-22:
If this is still relevant: building raw image (not ISO) and flashing that instead helps.