hamishcoleman/thinkpad-ec #250: L430

nanacrymi, 2023-12-02:
Hello hamishcoleman, Please can you send me the patch image for L430 so the can download it or assist me how I can do the procedure
Because I try and laptop list does not include my laptop  model

hamishcoleman, 2023-12-13:
Did you want to patch for the keyboard or the battery?  The L430 patch is not showing up on the laptop list because the way the keyboard patch was contributed doesnt follow the same way that the other patches work.

You should be able to configure for a battery patch and make the patched.l430.img as is, but the Makefile will need a small change if you are trying to patch for the keyboard layout.

leecher1337, 2024-02-04:
You can use this repository: https://github.com/leecher1337/thinkpad-ec
It contains a working patch infrastructure, just flashed another L430 including battery and keyboard patch today and it worked fine.
As we don't know which patches you want to apply (keyboard, bttery or both?), we cannot send you an .iso

Ensure that you update to the latest firmware version first, if you fail to do so because of battery error, just use patched https://github.com/hamishcoleman/thinkpad-ec/issues/203#issuecomment-1636705370 WINUPDP, I just verified that it works fine as expected.