hamishcoleman/thinkpad-ec #251: Overwritten bootloader

mirobartovic, 2023-12-10:
Hi, on my T530 when I do BIOS update using the official bootable CD image, it asks if I want to overwrite my custom bootloader (GRUB) or no. The patcher doesn't ask this and overwrites it automatically. The system is not bootable after patching and the bootloader has to be restored manually. I'm using Debian Bookworm, UEFI+SecureBoot enabled for standard system operation, Legacy for the patching.

hamishcoleman, 2023-12-13:
I've never seen the original bootable CD ask that question, nor have I seen any code in the installer software that looked like it is touching the MBR.

I'm honestly confused here, since the BIOS and EC updates dont have anything to do with the hard disk and booting, so I dont know why you would have had this overwritten.

Especially since you are using UEFI, which doesnt have the same fragile and easy to overwrite bootloader methods as the old MBR "active partitions" did.

mirobartovic, 2024-01-01:
I'm not really sure what happened there either. I was not expecting anything like that so maybe I don't remember the details correctly. But I'm sure the original CD asked me if I want to overwrite _something_, I declined and the laptop booted, while your patcher didn't ask and the laptop stopped booting. Nevertheless, repairing the boot process was simple and my new 3rd party battery is charging now. That is all I need, thank you very much for your hard work! I will close this but if I find myself really bored in the future, I will update the BIOS and EC again with both tools and document it. In case the problem shows up again, I will reopen this issue.

closed 2024-01-01 by mirobartovic