hamishcoleman/thinkpad-ec #252: X230: Using older BIOS (with same EC version) to remove genuine battery check but keep the WiFi whitelist disabled?

coaxial, 2023-12-12:
My X230 is running v2.73 (G2ETB3WW) which runs EC 1.14. The thinkpad-ec tool builds an image that contains v2.75 (G2ETB5WW) with EC 1.14.

If I flash the resulting artifact, will it upgrade my BIOS to v2.75?

If so, is there a way to use my existing v2.73 instead but the patched 1.14 EC to remove genuine battery checks? The reason being that my 2.73 BIOS has the WiFi whitelist removed, but the stock 2.75 will prevent me from using that WiFi card.

I have a dump of the modified 2.73 firmware, could I inject that into the ISO along with the patched EC 1.14?

hamishcoleman, 2023-12-13:
The patch is simply build using data from that BIOS version - it is the EC version that is important, and since that has not changed, the built image is compatible with your older BIOS version.  Additionally, the installer doesnt touch the BIOS, so there will be no reversions of your BIOS patches

coaxial, 2023-12-13:
Thank you for the information, closing as you've answered the question.

closed 2023-12-13 by coaxial