hamishcoleman/thinkpad-ec #255: Error 081 - Failed to load BIOS image file! Status = 14 // Lenovo B590

pacone97, 2024-02-24:
Hey,

I followed the steps and prepered my USB-Stick. When I boot from it, the information for the patch is displayed, it matches my Lenovo B590, and I'm prompted to press a key to continue.

If I do so, I do get the following output:

SecureFlash BIOS detected
Read BIOS image from file
Error 081 - Failed to load BIOS image file! Status = 14.

I tried using both repositories: 

https://github.com/hamishcoleman/thinkpad-ec
https://github.com/leecher1337/thinkpad-ec/tree/Lx30

I did try another USB-Stick aswell, that had no effect. Also I unmounted the Stick before flashing using umount. Did I miss anything?
I hope the provided information enables anyone to help me.

Also: I do get a error message saying the contents can't be displayed when I want to check its contents in Linux Mint.

Kind regards

hamishcoleman, 2024-03-18:
I dont actually have any experience with this style of system, so I dont know the usual flow for flashing the EC.  However, if you could share the log of your build, that will allow checking that what you built looks right.

12thDimension, 2025-05-06:
I got the same error but with status 81 instead. @hamishcoleman if you tell me what log you need / where to get it, Id gladly send it.