hamishcoleman/thinkpad-ec #257: reverse keyboard changes

antipodes5, 2024-05-28:
I have a standard x230 keyboard. I wanted to install an aftermarket battery

Since installing the patch, I have lost my 'Del' key. It now works as a 'PgUp'. 

(There may be a couple of other changes, but its the loss of the delete function that is killing me.)

How can I restore the normal function of the keyboard?

Lucaacer, 2024-05-28:
If you want to remove battery control without applying the keyboard patch, these are the commands you have got to use before building the img file

sudo make patch_disable_keyboard clean
sudo make patch_enable_battery clean

antipodes5, 2024-05-30:
Thanks, but that's not answering my question.

How do I get my original keyboard settings BACK?

Lucaacer, 2024-05-30:
You should apply the patch from yhe beginning. I mean, you have got to clone the repo, to choose the img file and, before building the img file you must follow this sequence

sudo make patch_disable_keyboard clean
sudo make patch_enable_battery clean
make patched.x230.img

Once the patched bios and EC patch img file have been built, just dd them on to the usb stick, boot you pc from the usb, flash the patched img and you should be ok.

Lucaacer, 2024-05-30:
Sorry, I suppose you'd better restore the official bios before applying the patch.