hamishcoleman/thinkpad-ec #261: E330 Linux erorr

iur4ic, 2024-08-24:
When I get to the parameter 
make patched.e330.img
Linux gives an error
Cannot read e330.H3EC35WW.img.d/001_keysym.patch
make: *** [Makefile:220: e330.H3EC35WW.img] Error 1

House-husband, 2024-09-11:
I have the same issue. Any chance someone can help please?

deeptenk, 2024-09-11:
> I have the same issue. Any chance someone can help please?
You probably need to update all linux packages. You can also download the patch. Sorry for my English.
(https://cloud.mail.ru/public/mCbZ/PA2G7cJzT)

House-husband, 2024-09-12:
> > Please forgive my lack of experience I am an absolute Noob and have ADHD as well. I have been fighting to do this mod to get rid of the non genuine battery issue for a few days now. I have learned a great deal and would have loved to have taken that hex code applied it myself and being able to reflash my bios with it. I have edited the hex code and saved the firmware file myself but it won't burn on to a disc as a bootable bios update. You have supplied these files, as I said I am a Noob but do want to learn, how do I use these files to get rid of this error and make the non genuine battery charge, please. Thank you
> 
> http://t.me/deeptenk I can help you, send me message.

Thank you deeptenk, I worked out that this was the end result and that it just needed putting onto a disc. I burned a copy and updated the bios+firmware, thus removing the problem. Thank you for you help, you have restored my faith in humanity, and saved me a lot of money!

deeptenk, 2024-09-12:
> Thank you deeptenk, I worked out that this was the end result and that it just needed putting onto a disc. I burned a copy and updated the bios+firmware, thus removing the problem. Thank you for you help, you have restored my faith in humanity, and saved me a lot of money!

No problem! You could simply create a bootable USB flash drive using the "Rufus" program and the *.img image. That would be even easier.