hamishcoleman/thinkpad-ec #266: X230 stuck on 2.75 - I cannot downgrade to 2.60!

brunocek, 2025-01-02:
I think I got myself in trouble stuck on 2.75 - I cannot downgrade to 2.60 (X230) via IVprep or the lenovo tool (I get "This app can't run on your PC"). Is there a way forward, please?

Here is what I did and was planning to do (1vyrain, skulls and clean/disable/break the Intel ME):

#### 1. EC Patching (accept all batteries and keyboards)
https://github.com/hamishcoleman/thinkpad-ec/
It recomends using on the last modifiable version: x230 -BIOS 2.75 (G2ETB5WW) EC 1.14 (G2HT35WW)

Therefore I upgraded from my 2.55 to 2.75, by downloading g2uj31us.exe, saving on a usb stick and running on the target machine's windows (the bootable iso would not dd well to the usb stick).
https://support.lenovo.com/ec/en/downloads/ds029187

#### 2. Downgrade from 2.75 to 2.60 (X230) (I cannot perform from here onwards):
https://github.com/n4ru/IVprep/tree/master

Downloaded both IVprep and g2uj17us.exe and saved on the usb stick to run on target machine's windows but then I get I get "This app can't run on your PC".

I have tested with both:
BIOS Setup -> Security -> UEFI BIOS Update Option. Set Flash BIOS Updating by End-Users to Enabled, and Secure RollBack Prevention to Disabled
BIOS Setup -> Security -> UEFI BIOS Update Option. Set Flash BIOS Updating by End-Users to Enabled, and Secure RollBack Prevention to Enabled

#### 3. Get rid of the ME
https://github.com/corna/me_cleaner
I gues for this one, I should boot the target machine with a Live Linux, clone the repo and execute

#### 4. Install Skulls via 1vyrain
https://github.com/n4ru/1vyrain

This would be burn the usb stick to the 1vyrain, use the ethernet cable and the skulls release URL
https://github.com/merge/skulls/releases

## This question has been also posted on:
- https://github.com/n4ru/1vyrain/issues/144
- https://forum.qubes-os.org/t/how-to-use-ivprep-and-1vyrain-and-disable-intel-me-on-a-lenovo-x230/18162/5
- https://github.com/merge/skulls/issues/302

brunocek, 2025-05-19:
I tried downgrading the BIOS via a reputable method ([GitHub - gch1p/thinkpad-bios-software-flashing-guide: flashing coreboot on thinkpads without external programmer](https://github.com/gch1p/thinkpad-bios-software-flashing-guide#downgrading-bios)). But I got:

> ERROR 081 - Failed to load BIOS image file! Status = 14.

It booted and it tried to run `autoexec.bat` gave me this error and a DOS prompt. I tried again running:

`dosflash.exe /sd /file G2ETB3WW\$01D3000.FL1`

And I got the same error…

Therefore in my case: X230 - when EC Pathcing with thinkpad-ec - one must be very cautious not to upgrade to 2.75 as thinkpad-ec recommends…

Any ideas of what I could do to solve it with external (usb) flashing?

Will I need to internal flash with my hardware programmer? - I am enroute to this whilst I read more and more. Thank you for any help or clarification in advance!

hacshyac, 2025-08-27:
I was in the same case like you (from 2.75 to 2.60), the ERROR 081, but I checked again and found that I did not change the directory'name correctly.

I wrote this when I got ERROR 081 (it is the directory's name on the tutorial page [https://github.com/gch1p/thinkpad-bios-software-flashing-guide#bios-versions](url), I only changed $01D2000.FL1 to $01D3000.FL1 but forgot to change the name before) `dosflash.exe /sd /file G1ET93WW\$01D3000.FL1`

Then I changed to the right name and I can run the image via usb-stick smoothly. `dosflash.exe /sd /file G2ETB3WW\$01D3000.FL1`