hamishcoleman/thinkpad-ec #268: How to add patch when one iso contains 2 different firmware versions?

leecher1337, 2025-02-11:
I want to submit a Patch for X130e thinkpad.
x130e comes in 2 flavous, one with AMD and one with Intel CPU. For both flavours, BIOS updates are on the same ISO.

So I added:
```
x131e.G8HT52WW.s0AG8000.FL1 rule:FL2,dep:g8uj31us.iso,depi:g8uj31us.iso.bat1,param:0AG8000.FL1 x131e g8 BIOS 2.98 Flash File
x131e.G9HT51WW.s0ag9000.FL1 rule:FL2,dep:g8uj31us.iso,depi:g8uj31us.iso.bat1,param:0ag9000.fl1 x131e g9 BIOS 2.98 Flash File

x131e.G8HT52WW.img    rule:IMGnoenc,dep:x131e.G8HT52WW.s0AG8000.FL1 x131e EC 2.02 (not encrypted) (8085 cpu)
x131e.G9HT51WW.img    rule:IMGnoenc,dep:x131e.G9HT51WW.s0ag9000.FL1 x131e EC 2.02 (not encrypted) (8085 cpu)

patched.x131e.iso  rule:niceISO,dep:g8uj31us.iso,suffix:0,insert:0  for patching Thinkpad X131e
```

But this would naturally create one ISO file with only the last one of the 2 patches instead of 2 ISOs with the respective number (G8HT52WW,G9HT51WW) .

The patching infrastructure is largely undocumented, so please advise how to add the patch correctly.
Thanks.

hamishcoleman, 2025-02-11:
The patching rules were written with the assumption that a single product code referred to a single product.

Since you have found one ISO product code that has multiple products inside it, we are a victim of marketing trumping technology :-(

One helpful thing for extraction is that the g9 FL1 file does have a different product code /inside/ the ISO file ("G9HT52WW")

So, instead of trying to force two firmware files into the one ISO, we could pretend that the ISO file was called "G9...." all along and end up with a g8 /and/ a g9 ISO file. (The alternative would need updates to the generate deps script and a new Makefile rule macro written)

Starting with your lines above, I got the following patch before realising that there was no UEFI capsule handling code in the FL2_copyIMG script

```diff
diff --git a/Descriptions.txt b/Descriptions.txt
index 75fb361e9066..e818dd6100a9 100644
--- a/Descriptions.txt
+++ b/Descriptions.txt
@@ -260,7 +260,7 @@ t530.G4HT40WW.s01D5100.FL2  rule:FL2,dep:g4uj41us.iso,depi:g4uj41us.iso.bat,para
 w530.G4HT39WW.s01D5200.FL2  rule:FL2,dep:g5uj37us.iso,depi:g5uj37us.iso.bat,param:01D5200.FL2  w530
 w530.G4HT40WW.s01D5200.FL2  rule:FL2,dep:g5uj39us.iso,depi:g5uj39us.iso.bat,param:01D5200.FL2  w530
 x131e.G8HT52WW.s0AG8000.FL1 rule:FL2,dep:g8uj31us.iso,param:0AG8000.FL1                        x131
-#x131e.G8HT52WW.s0ag9000.FL1 rule:FL2,dep:g8uj31us.iso,param:0ag9000.FL1                        x13
+x131e.G9HT52WW.s0ag9000.fl1 rule:FL2,dep:g8uj31us.iso,param:0ag9000.fl1,insert:0               x131
 x1cg1.G6HT24WW.s01D7000.FL2 rule:FL2,dep:g6uj24us.iso,param:01D7000.FL2                        x1c 
 x200.7XHT22WW.s01B9000.FL2  rule:FL2multi2,dep:6duj37uc.iso,depi:6duj37uc.iso.bat,param:01B9000.FL2sh File
 x200.7XHT24WW.s01B9000.FL2  rule:FL2multi2,dep:6duj47uc.iso,depi:6duj47uc.iso.bat,param:01B9000.FL2sh File
@@ -316,6 +316,8 @@ t530.G4HT39WW.img     rule:IMG,dep:t530.G4HT39WW.s01D5100.FL2       t530 EC 1.13
 t530.G4HT40WW.img     rule:IMG,dep:t530.G4HT40WW.s01D5100.FL2       t530 EC 1.14 (decrypted) (ARCom
 w530.G4HT39WW.img     rule:IMG,dep:w530.G4HT39WW.s01D5200.FL2       w530 EC 1.13 (decrypted) (ARCom
 w530.G4HT40WW.img     rule:IMG,dep:w530.G4HT40WW.s01D5200.FL2       w530 EC 1.14 (decrypted) (ARCom
+x131e.G8HT52WW.img    rule:IMGnoenc,dep:x131e.G8HT52WW.s0AG8000.FL1 x131e EC 2.02 (not encrypted) (
+x131e.G9HT52WW.img    rule:IMGnoenc,dep:x131e.G9HT52WW.s0ag9000.fl1 x131e G9 EC 2.02 (not encrypted
 x1cg1.G6HT24WW.img    rule:IMG,dep:x1cg1.G6HT24WW.s01D7000.FL2      x1c Gen 1 EC 1.06 (decrypted) (
 x200.7XHT22WW.img     rule:IMGnoenc,dep:x200.7XHT22WW.s01B9000.FL2  x200 EC 1.04 (not encrypted) (H
 x200.7XHT24WW.img     rule:IMGnoenc,dep:x200.7XHT24WW.s01B9000.FL2  x200 EC 1.06 (not encrypted) (H
```

leecher1337, 2025-02-11:
I got around the issue by using 2 different ISOs, which thankfully is no problem for this particular case, but then I got struck by the .bat1 handler, as it always assumes that the first .FL1 file in the image is the one to patch. I guess we need a rule with param: to do the patching instead, which would be a cleaner approach anyway, but I'm not particulary good with Makefiles, unfortunately.
Any clues for that one?

As for the UEFI capsule handler, it simply is:
```patch
diff --git a/scripts/FL2_copyIMG b/scripts/FL2_copyIMG
index 95804b7..cc9acf6 100755
--- a/scripts/FL2_copyIMG
+++ b/scripts/FL2_copyIMG
@@ -615,6 +615,7 @@ sub _check {
          9437264 => 1,  # l430
         13631568 => 1,  # e330
         12587008 => 1,
+        13107280 => 1   # x131e
     };

     my $capsule_offset_hack = $self->{capsule_offset_hack} || 0;
@@ -686,7 +687,7 @@ sub _check {

         push @{$self->{header}{dir}}, $dir;

-        if ($name eq 'Ec') {
+        if (lc($name) eq 'ec') {
             $self->set_offset_size(
                 $dir->{FileOffset}+$capsule_offset_hack,
                 $dir->{Size},
```