hamishcoleman/thinkpad-ec #269: support for battery issue of thinkpad edge e520

renos1983, 2025-02-23:
Hi guys, is there a possibility of a battery patch for the lenovo thinkpad edge e520? Unfortunately I bought a conventional battery and my laptop is showing the familiar message that the battery is not genuine.

hamishcoleman, 2025-02-24:
Sadly, it is not simple.

I had a quick look and this hardware is using a different ISO layout with a different FL1 file format and no obvious commonality with the EC that we know how to patch.  The flash tool also seems to be different.

All that adds up to a lot of unknowns that have no simple answers

renos1983, 2025-02-24:
Thank you very much for the prompt response to my question. That's what I understood, that probably nothing can be done and I will have to return the battery back to the store. Thanks again.

leecher1337, 2025-12-28:
The check routine in BIOS is pretty similar to the one of the Lx30 series described [here](https://github.com/leecher1337/thinkpad-Lx30-ec/blob/main/doc/battery.md),
but with a different command (0x53):

```c

if ((EcIoDxe->Command(EcIoDxe, 0x88) & 1) != 0)
{
	if ((EcIoDxe->Command(EcIoDxe, 0x53) & 0x20) == 0)
	{
		// Output:       L"This system does not support batteries that are not genuine Lenovo-made or\n"...
	}
}
```

It's also a NUVOTON KBC (CR16 CPU) as we already know it from the Lx30 series.
My guess is that the battery state machine dispatcher is at 27dea in EC code,
but state machine is different to the one present in the Thinkpad Lx30 series.
The routine to check for the expected 0x53 variable is at function 26f20,
if you ask me, look at case 3 at 270a0. (0x50 + 3 = 0x53).
These offsets are for the firmware present in $0A8H000.FL1 
This may serve you as a starting point to find and patch the battery state machine.